xmos/lib_xud #141: An XTAG4 device address cannot be set if a powered hub is unplugged from the upstream PC and re-plugged (maintaining power to the XTAG4 contionuously)

MarkPhillips-xmos, 2021-04-22:
An XTAG4 is plugged into a powered up and is running its downloaded firmware (delivered in Tools 15.0.5). The hub is kept powered and unplugged from the host PC. It is plugged back into the host PC. Its is not enumerated correctly. The dmesg report  suggests the set address fails.

This seems similar in nature to issue reported by Ed (#124)

 
*** Plugged in XTAG 
[  +0.491629] usb 1-2.4: new high-speed USB device number 6 using xhci_hcd
[  +0.084480] usb 1-2.4: New USB device found, idVendor=20b1, idProduct=f7d5
[  +0.000013] usb 1-2.4: New USB device strings: Mfr=1, Product=2, SerialNumber=5
[  +0.000008] usb 1-2.4: Product: XMOS XTAG-4
[  +0.000007] usb 1-2.4: Manufacturer: XMOS
[  +0.000007] usb 1-2.4: SerialNumber: XD00FTAGpr1V0_15
 
*** Unplugged hub
 
[Apr22 11:51] usb 1-2: USB disconnect, device number 3
[  +0.000013] usb 1-2.1: USB disconnect, device number 4
[  +0.000979] usb 1-2.4: USB disconnect, device number 6
 
 
*** plugged hub in
 
[ +28.494170] usb 1-2: new high-speed USB device number 7 using xhci_hcd
[  +0.123206] usb 1-2: New USB device found, idVendor=1a40, idProduct=0101
[  +0.000014] usb 1-2: New USB device strings: Mfr=0, Product=1, SerialNumber=0
[  +0.000008] usb 1-2: Product: USB 2.0 Hub
[  +0.000879] hub 1-2:1.0: USB hub found
[  +0.000071] hub 1-2:1.0: 4 ports detected
[  +0.273588] usb 1-2.1: new high-speed USB device number 8 using xhci_hcd
[  +0.084423] usb 1-2.1: New USB device found, idVendor=1a40, idProduct=0101
[  +0.000011] usb 1-2.1: New USB device strings: Mfr=0, Product=1, SerialNumber=0
[  +0.000007] usb 1-2.1: Product: USB 2.0 Hub
[  +0.001985] hub 1-2.1:1.0: USB hub found
[  +0.000052] hub 1-2.1:1.0: 4 ports detected
[  +0.073718] usb 1-2.4: new high-speed USB device number 9 using xhci_hcd
[  +0.072912] usb 1-2.4: Device not responding to setup address.
[  +0.201248] usb 1-2.4: Device not responding to setup address.
[  +0.200867] usb 1-2.4: device not accepting address 9, error -71
[  +0.072977] usb 1-2.4: new high-speed USB device number 10 using xhci_hcd
[  +0.687068] usb 1-2.4: device descriptor read/64, error -71
[  +0.588005] usb 1-2.4: device descriptor read/64, error -71
[  +0.100889] usb 1-2-port4: attempt power cycle
[  +0.777018] usb 1-2.4: new high-speed USB device number 11 using xhci_hcd
[  +0.000171] usb 1-2.4: Device not responding to setup address.
[  +0.201014] usb 1-2.4: Device not responding to setup address.
[  +0.200829] usb 1-2.4: device not accepting address 11, error -71
[  +0.072998] usb 1-2.4: new high-speed USB device number 12 using xhci_hcd
[  +0.000164] usb 1-2.4: Device not responding to setup address.
[  +0.200996] usb 1-2.4: Device not responding to setup address.
[  +0.200833] usb 1-2.4: device not accepting address 12, error -71
[  +0.000113] usb 1-2-port4: unable to enumerate USB device
 

Hub is a Belkin:
[  +0.000064] hub 1-2:1.0: 4 ports detected
[  +0.273301] usb 1-2.1: new high-speed USB device number 64 using xhci_hcd
[  +0.084396] usb 1-2.1: New USB device found, idVendor=1a40, idProduct=0101
[  +0.000011] usb 1-2.1: New USB device strings: Mfr=0, Product=1, SerialNumber=0
[  +0.000008] usb 1-2.1: Product: USB 2.0 Hub
[  +0.000719] hub 1-2.1:1.0: USB hub found
[  +0.000063] hub 1-2.1:1.0: 4 ports detected

[image]

MarkPhillips-xmos, 2021-04-23:
Note we also see this issue with the XTAG3 (in fact it was first discovered by the Si team with XTAG3s).

xross, 2021-06-21:
A fix went related to signalling went in here: https://github.com/xmos/lib_xud/pull/157 I would be interesting to know if this helps the issue at all.

MarkPhillips-xmos, 2021-06-21:
A manual test of the branch DEV/XS3 (68d501450a0562a00c841b05b8da75860b70e542) on a setup using a the Belkin hub plus a "DELL docking station" shows the issue to be fixed. More extensive tests will be carried out with combinations of equipment as they becomes available.

MarkPhillips-xmos, 2021-06-21:
Hi Ross,

As commented the fix looks good on the kit I have here on which to test.

Is there any chance this will be fixed for the XTAG3 (XS1) device – the issue was originally observed with a XTAG3 (by Nick in the lab)?
For XTAG3 we are using the following USB repos and tags:

sh 'git clone git://git/apps/sc_usb .'
sh 'git checkout v1.0.3rc0'
sh 'git clone git://github.com/xcore/sc_usb_device .'
sh 'git checkout v1.3.2rc0'
sh 'git clone git://git/apps/sc_xud .'
sh 'git checkout v2.2.1rc0'
sh 'git clone git://github.com/xcore/sc_util .'

Cheers
Mark

xross, 2021-06-24:
Thanks for the feedback @MarkPhillips-xmos. 

We've actually stripped out XS1 support from lib_xud now. The code is (was) quite different for XS1 so i'm not sure this fix would apply.

xross, 2022-01-21:
Believe to be resolved.

closed 2022-01-21 by xross