xmos/lib_xud #162: Project files out of date

xross, 2021-05-25:
The project files have not been maintained as version number incremented

mbanth, 2021-05-25:
Eamail from @xross providing further detail:

Hi Chaps, 

Not sure quite who this sit under these days…

The project files for xTimeComposer (XTC) don’t seem to be being maintained when libs are released.

Specially the version numbers in .xproject don’t match module_build_info, CHANGELOG.rst or the git tags.

This means that when someone imports a lib into XTC they see the wrong version number, quite confusing.

We still have customers and FAE’s using XTC (that’s how I noticed this issue) so I think this needs addressing.

My memory says that the updating of project files was somewhat automated and a script needed running before each release.

Cheers
Ross

xross, 2022-07-27:
Project files deleted.

closed 2022-07-27 by xross