xmos/lib_xud #172: Error while running DEV/XS3 branch with XVF3000

lucianomartin, 2021-06-14:
I included the DEV/XS3 branch in the view sw_vocalfusion_new_xud.txt and I tested the XVF3000 build 1i2o2_lin33_usbctl. If I use "xrun --xscope", when I try to record some audio with Audacity I get this error:

```
xrun --xscope bin/1i2o2_lin33_usbctl/__app_vf_speaker_base_1i2o2_lin33_usbctl.xe 
WARNING: tile[0] - xSCOPE not supported by application, I/O will be via JTAG
WARNING: tile[1] - xSCOPE not supported by application, I/O will be via JTAG

xrun: Program received signal ET_ILLEGAL_RESOURCE, Resource exception.
      [Switching to tile[1] core[4] (dual issue)]
      0x000465d0 in Pid_In ()
```

When I use xflash I don't get the same error when I try to record with Audacity, but Audacity doesn't record 
either. The dumpstate is attached.
[dumpstate_xvf3000.txt](https://github.com/xmos/lib_xud/files/6649222/dumpstate_xvf3000.txt)

xross, 2021-06-14:
lib_xud traps with Exception Type 0x14  on an IN instruction supposedly on a port but the Exception Data is 0x6f814 (456724)
i.e. not a valid resource ID. My first guess would be that you have to memory corruption - doesn't look like an issue with lib_xud on the face of it

lucianomartin, 2021-06-15:
Thanks for the analysis. I am not sure what can cause the exception. lib_xud has most of the changes, while in lib_xua we updated the timers:
https://github.com/xmos/lib_xua/compare/develop...mbanth:feature/align_with_XUD?expand=1
and in sw_vocalfusion we just updated some flags and the XN file to support tools 15:
https://github.com/xmos/sw_vocalfusion/compare/develop...feature/align_with_XUD

Do you have any suggestion about how I can debug this exception?

lucianomartin, 2021-06-17:
This is another dumpstate, but in a slightly different place for decouple.xc.
[dumpstate_xvf3000_1.txt](https://github.com/xmos/lib_xud/files/6669429/dumpstate_xvf3000_1.txt)

xross, 2021-06-17:
> I have reproduced a very similar problem on Windows with an XVF3510. Playing audio to the device triggers the resource exception. A copy of the command line, the dump state and a Beagle 480 USB trace are in the zip file: [illegal-resource-exception-2021-06-17-15-10.zip](https://github.com/xmos/lib_xud/files/6670950/illegal-resource-exception-2021-06-17-15-10.zip).
> 
> The XVF3510 build uses the sw_xvf3510_develop view file as of 15:10 on 2021-06-17 with one exception, the sw_xvf3510 repository uses the HEAD of the _feature/align_with_XUD_ branch instead of the HEAD of _develop_.

Is this running on xs2 or xs3 please?

mbanth, 2021-06-17:
> Is this running on xs2 or xs3 please?

It's running on an XS2 part, the one used for XVF3510.

xross, 2021-06-17:
Out of interest, what happens when the order of the main par{} is modified on the USB tile?

xross, 2021-06-17:
> I have reproduced a very similar problem on Windows with an XVF3510. Playing audio to the device triggers the resource exception. A copy of the command line, the dump state and a Beagle 480 USB [...]

This looks like very similar memory corruption (0x615f0 used as a port ID). Which HW do I need to reproduce this issue?

mbanth, 2021-06-17:
I have reporduced the problem using the last public release of XVF3510 with only two changes:
- Using the HEAD of the _DEV/XS3_ branch in lib_xud, and
- Removing the three arguments from the call to `XUD_Main()` that existed in previous versions of lib_xud but have been removed from the current version.

Because it uses the most recent public release of XVF3510, the executable does not include any of the changes made to support Configurable HID, e.g., those in lib_xua.

Details in [illegal-resource-exception-2021-06-17-17-01.zip](https://github.com/xmos/lib_xud/files/6671649/illegal-resource-exception-2021-06-17-17-01.zip).

mbanth, 2021-06-17:
> Which HW do I need to reproduce this issue?

An XVF3510 dev kit.  I have sent some details to @xross directly in an email.

mbanth, 2021-06-18:
I have verified that the most recent XVF3510 public release, without modifications, does not exhibit this issue.  This test, coupled with the experiment using the _DEV/XS3_ branch of lib_xud and the slightly modified call to `XUD_Main()`, indicates that the changes between the HEAD of _develop_ and the HEAD of _DEV/XS3_ have either:
- unmasked an existing bug somewhere in the XVF3510 firmware, or
- introduced a bug in XUD.

Only further investigation will reveal which has happened.

mbanth, 2021-06-18:
I have performed two experiments that re-order the call to `XUD_Main()` in the `par` statement that forks USB logical cores.  In both experiments, the illegal resource exception still occurs in the XUD logical core.  From what I can see, the exception signature has not changed except for the ID number of the logical core in which it occurs.

Experiment 1 details appear in [illegal-resource-exception-2021-06-18-14-35.zip](https://github.com/xmos/lib_xud/files/6677602/illegal-resource-exception-2021-06-18-14-35.zip).  Experiment 2 details appear in [illegal-resource-exception-2021-06-18-15-07.zip](https://github.com/xmos/lib_xud/files/6677609/illegal-resource-exception-2021-06-18-15-07.zip).

mbanth, 2021-06-23:
Hopefully the steps below will reproduce the issue Luciano and I have seen on XVF3000 and XVF3510:
1. Set up the XVF3510 board, information which you may find helpful
    1. XM-014272-PC, XVF3510 Dev Kit Setup Guides, however
        1. You do not need to have operational speakers to reproduce the issue
        1. You will need to have an xTag plugged in and connected to your laptop
    1. XM-014232-PC, XVF3510 User Guides, Draft 5a, Section 3.6, pp 16-18
    1. XM-014236-DA, XVF3510 Development Kit Design Advisory if you received a 1v0 or 1v1 board
    1. The Dev Kit is more limited in functionality than you may expect.  Specifically, the 3.5 mm jack on the main board is a Line In jack, not a Line Out, and the Line Out jack on the auxiliary board will not work without having a Raspberry Pi installed underneath it.  Do not install a Raspberry Pi, you will not need it and configuring it to work can take quite a bit of time.
    1. Only supply power to the main Dev Kit board, not the auxiliary board.
2.  Create a sandbox using the sw_xvf3510_master view file and view.py
3.  Follow [these instructions](https://xmosjira.atlassian.net/wiki/spaces/HYD/pages/1375076387/How+to+build+the+3510+firmware+on+WSL) to build the code.  Some of the instruction steps may be unnecessary depending on what you already have installed on your machine.  You have to have a virtual environment with python 3.7.6 active to build successfully.
4.  Once you can build the stock code, change lib_xud to use the HEAD of the _DEV/XS3_ branch instead of the HEAD of _master_
5.  Change the call to `XUD_Main()` in sw_xvf3510/app_xk_xvf3510_l71/src/usb_wrapper.xc to remove the three arguments no longer supported in XUD, i.e., “null, null, -1”.
6.  Build again, i.e., from sw_xvf3510/app_xk_xvf3510_l71 run `waf clean build –config=usb_adaptive`
7.  Run the code on the XVF3510 board with `xrun –xscope-io-only app_xk_xvf3510_l71_usb_adaptive.xe`
8.  Use Audacity to play a stereo track to the XVF3510 at 48 kHz.  I used //projects/hydra_audio/acoustic_team_test_audio/playback_audio/rick_stereo.wav.  It doesn’t matter if Audacity is configured to listen to the XVF3510 microphones or not.

The Resource exception happens very quickly after starting the streaming audio.  Usually, I see the exception on the call to `Pid_Sof()`.  However, I have once seen it and Luciano has reported it on the call to `Pid_In()`.  As noted in previous comments, the exception does follow XUD when swapping the order of statements within the `par`.

xross, 2021-06-23:
The package  containing the XVF3510 board has arrived with me today.

xross, 2021-06-24:
What am I doing wrong please?

```
(.venv) 
ross@muc ~/sandboxes/sb_3510/sw_xvf3510 (master)
$ waf configure clean build --config=usb_adaptive
develop soon
```

xross, 2021-06-24:
Bisecting, the issue appears occurs between the two commits:

095f6d2 (Feb 26th) and 00e91ee (Feb 18th)

xross, 2021-06-28:
Genuine issue in lib_xud where adding dual issue and XS3 support had broken XS2 support in some instances.

xross, 2021-06-28:
Confirmed resolves issue with xvf3510.

closed 2021-06-28 by xross