xmos/lib_xud #188: Formatting lost messages on build fails from pytest

xross, 2021-06-23:
Formatting broken/string not handled properly. For example:

ERROR: build failed.
b'Checking build modules\n'b'Using build modules: lib_xud(2.0.0)\n'b'Analyzing main.xc\n'b'In file included from ../src/main.xc:4:\n'b'In file included from ../../shared_src/shared.h:10:\n'b'/Users/ross/sandboxes/sb_xud_view/lib_xud/lib_xud/api/xud.h:42:2: warning: XUD_CORE_CLOCK not defined, using default (700MHz)\n'b'#warning XUD_CORE_CLOCK not defined, using default (700MHz)\n'b' ^\n'b'In file included from ../src/main.xc:4:\n'b'In file included from ../../shared_src/shared.h:10:\n'b'/Users/ross/sandboxes/sb_xud_view/lib_xud/lib_xud/api/xud.h:42:2: warning: XUD_CORE_CLOCK not defined, using default (700MHz)\n'b'#warning XUD_CORE_CLOCK not defined, using default (700MHz)\n'b' ^\n'b'In file included from ../src/main.xc:4:\n'b'../../shared_src/shared.h:132:1: warning: control reaches end of non-void function [-Wreturn-type]\n'b'}\n'b'^\n'b"../src/main.xc:17:26: error: use of undeclared identifer `XUD_EP_COUNT_IN'\n"b'XUD_EpType epTypeTableIn[XUD_EP_COUNT_IN] =   {XUD_EPTYPE_CTL, \n'b'                         ^\n'b'../src/main.xc:17:26: error: size of array is not constant or based on a const static parameter\n'b'XUD_EpType epTypeTableIn[XUD_EP_COUNT_IN] =   {XUD_EPTYPE_CTL, \n'b'                         ^~~~~~~~~~~~~~~\n'b'../src/main.xc:54:1: warning: control reaches end of non-void function [-Wreturn-type]\n'b'}\n'b'^\n'b"../src/main.xc:27:18: warning: unused variable `res' [-Wunused-variable]\n"b'    XUD_Result_t res;\n'b'                 ^~~\n'b"../src/main.xc:31:12: warning: unused variable `ep_in' [-Wunused-variable]\n"b'    XUD_ep ep_in  = XUD_InitEp(c_in);\n'b'           ^~~~~\n'b'xmake[1]: *** [.build_xs3/src/main.xc.pca.xml.decouple] Error 1\n'b'xmake: *** [analyze] Error 2\n'

xross, 2021-06-29:
In the latest code version build fails do not make to to the terminal 

_____________________________________________________________ test_RunUsbSession[xs3-1-0-HS] ______________________________________________________________

test_session = <usb_session.UsbSession object at 0x10f0197d0>, arch = 'xs3', ep = 1, address = 0, bus_speed = 'HS'
test_file = '/Users/ross/sandboxes/sb_xud_view/lib_xud/tests/test_bulk_tx_basic.py', capfd = <_pytest.capture.CaptureFixture object at 0x10efae290>

    def test_RunUsbSession(test_session, arch, ep, address, bus_speed, test_file, capfd):
    
        tester_list = []
        output = []
        testname, extension = os.path.splitext(os.path.basename(__file__))
        seed = random.randint(0, sys.maxsize)
    
        (clk_60, usb_phy) = get_usb_clk_phy(verbose=False, arch=arch)
        tester_list.extend(
            do_usb_test(
                arch,
                ep,
                address,
                bus_speed,
                clk_60,
                usb_phy,
                [test_session],
                test_file,
                seed,
            )
        )
        cap_output, err = capfd.readouterr()
        output.append(cap_output.split("\n"))
    
        sys.stdout.write("\n")
        results = Pyxsim.run_tester(output, tester_list)
    
        # TODO only one result
        for result in results:
            if not result:
                print(cap_output)
>           assert result
E           assert False

conftest.py:103: AssertionError

kevinyeungxmos, 2021-06-29:
see #200

closed 2021-07-05 by xross