xmos/lib_xud #212: __app_test_mode readme contains several errors

DaveAtkinsonXMOS, 2021-07-01:
On the DEV/XS3 branch (at the time of writing) there exists a readme file. The first line says
> This application stand-alone binaries for USB test modes. There are four build configurations, one for each USB test modes

This should probably read

> **These applications are** stand-alone binaries to test the USB test modes. There are four build configurations, one for each USB test mode.

But I don't know, so I won't just change it without someone vetting that. Also the final paragraph
> The application enters it's respective test mode from boot, this removing the requirement to set the mode from a host via the USBHSETT tool.
> 
> Note, you not shouild expect the the device to appear on any USB bus and its probabky not advisable to plug into any standard host.
> 
> This binaries are most commonly used in device characterisation.

Should probably read
> The **applications enter their** respective test mode from boot, **thus** removing the requirement to set the mode from a host via the USBHSETT tool.
> 
> Note, you **should not** expect the the device to appear on any USB bus and **it's probably** not advisable to plug **it** into any standard host.
> 
> **These** binaries are most commonly used in device characterisation.

I've marked the proposed changes in bold.

xross, 2021-07-01:
All suggested changed look good to me

xross, 2021-07-02:
Should be fixed now.

closed 2021-07-02 by xross