xmos/lib_xud #227: USB comms broken by commit Opts to XUD_Token_In_DI. Speed improvements to stall handling 933dda1

DaveAtkinsonXMOS, 2021-07-12:
The two lib_xud example tested (AN00124 and AN00129) work up to this [commit](https://github.com/xmos/lib_xud/commit/933dda1c1975a09871bc52813de10078295219b3), and then fail afterwards.

DaveAtkinsonXMOS, 2021-07-12:
Here is a Total Phase Data Center trace from a Beagle, xcore.ai running AN00129 and failing.
[AN000129 Linux Failure xcore.ai.tdc.zip](https://github.com/xmos/lib_xud/files/6800447/AN000129.Linux.Failure.xcore.ai.tdc.zip)

xross, 2021-07-12:
It looks like after a EP0 request that is halted (Set Idle) the IN Ep0 never seems to be marked ready again (NAKs forever)

xross, 2021-07-12:
The spec states that an EP that receives a SETUP request should be un-stalled. These changes break this functionality

commit 58b7aeff (2021-07-12): Merge pull request #228 from xross/fix/227

Fix/227 which closes #227
closed 2021-07-12 by DaveAtkinsonXMOS