xmos/lib_xud #244: Faster than 100MHz ref clock causes failure to enumerate on HID example running FS in XS3

ed-xmos, 2021-07-16:
This is causing an issue for 3610 where we have 540.6666 MHz core resulting in 108.1333 MHz ref clock. If we set ref clock to 90.111 MHz then it works.

To reproduce this, set the XN core clock in the HID example to 540 and then flip the ref clock between 108 and 90. 90 works, 108 does not.

xross, 2021-07-19:
The failure isn't just a function of reference clock since 600/120MHz operates as expected

ed-xmos, 2021-07-20:
My workaround to use a 90MHz ref clock in the 3610_UA works most of the time but reboot intensive tests such as test_dfu_UA which does 30+ reboots usually fails. I caught the failures and each case is a failure to enumerate (on the troublesome hub) - the classic no ACK to set address. So this solution is not sufficiently robust for the 3610_UA product

xross, 2021-07-22:
Experiments:
- AN00129
- MacBook Pro (2016) 
- HEAD of develop (3d0bfb33154038d09b7b1edb05196c6ad537af49 )

- HS, 540/108: Fails first SETUP transaction, back ACK with additional byte (0xD2FE, 0xD2D2 etc)
- FS, 540/108: Seems to operate as expected. Possibly slightly unreliable enumeration

Suspect above causes by bad sync to 60MHz USB clock. Moved to Eds' recommended settings of
1, 0, 1, 1 for RX_RISE_DELAY | RX_FALL_DELAY | TX_RISE_DELAY | TX_FALL_DELAY.

This resolved HS issue on my setup.

xross, 2021-07-22:
Next steps:
- Insert hub into chain and retry
- Run sim tests @ 540MHz

xross, 2021-07-22:
sim tests @ 540/108MHz: interestingly, on the latest codebase, only the following tests now  fail:
`
============================================================================== short test summary info ===============================================================================
FAILED test_ping_stall.py::test_RunUsbSession[xs3-1-1-HS-6-540] - assert False
FAILED test_suspend_resume.py::test_RunUsbSession[xs3-1-1-HS-6-540] - assert False
================================================================ 2 failed, 27 passed, 2 xfailed in 486.61s (0:08:06) ================================================================
'

Note, @500MHz many test failures are experienced so its reaching the edge (all pass at 600MHz)

xross, 2021-07-22:
Hub in chain.

Hub model: Belkin F5U233 (very short cables used)

1 hub:
This causes various packet errors on bus, though does enumerate and function.

2 hubs:
Yields similar results

Current conclusion is we are simply running out of MIPS. As you gradually reduced the core freq you can see the occurrence of errors increasing.  At the moment we should not recommend running lib_xud on XS3 < 600MHz.

Note, XS3 was taped out to support USB at 700MHz so we have already had to squeeze out extra performance.

xross, 2021-07-22:
Additionally, the current code base has some bug-fixes regarding stalling that could well impact performance. These relate to HALTing endpoints. This must be cleared on a SETUP transaction, which is where we see most of the packet errors. 

We should probably also look into why more sim tests don't fail at 540MHz.

xross, 2021-07-22:
As an experiment I tried removing some instructions such as CRC checking, clearing STALL etc and the packet errors disappear, this further supports my hypothesis.

closed 2021-07-27 by xross