xmos/lib_xud #247: Evaluate CPM for resolving dependancies

xross, 2021-07-21:
The current dependancy management system is non-ideal. 

For instance, viewfiles are not available to external users.

Current proposal is to use cmake cpm

DaveAtkinsonXMOS, 2021-08-10:
Started to investigate the [sw_hellocpm](https://github.com/keithm-xmos/sw_hellocpm) repository. This is a good starting point, but it will not suffice to build XC file without having to set explicit build instructions for each XC file. I consider a better solution is to create a new 'XC' language for CMake. I have created a first attempt at this, and I am trying to get it accepted as a merge request on CMake. Currently CMake want an XMOS C and C++ platform created first before considering a new language, so I am engaged in that. All work for this can be found at https://gitlab.kitware.com/DaveAtkinsonXMOS

DaveAtkinsonXMOS, 2021-08-13:
Currently trying to mediate between CMake and XMOS compiler devs to come up with a way of programmatically identifying the compilers...

xross, 2022-10-19:
Likely we're moving to a West based system. Regardless this issue is wider that just this repo. Closing.

closed 2022-10-19 by xross