xmos/lib_xud #287: Cannot run tests due to xcov addtions

xross, 2021-11-03:
I get the following error:

$ pytest --smoke
ImportError while loading conftest '/Users/ross/sandboxes/sb_xud_view2/lib_xud/tests/conftest.py'.
tests/conftest.py:20: in <module>
    combine_test = combine_process(os.path.dirname(os.path.abspath(__file__)))
../Installs/Mac/lib/python/xcoverage/xcov.py:659: in __init__
    self.resufd = open("%s/result/xcoverage_result.txt" % testpath, "w")
E   FileNotFoundError: [Errno 2] No such file or directory: '/Users/ross/sandboxes/sb_xud_view2/lib_xud/tests/result/xcoverage_result.txt'

Also, psutil needs adding to requirements.txt

kevinyeungxmos, 2021-11-03:
Set env first. In infr_scripts_pl:
`source ./SetupEnv`
Then, run Build.pl in test_support:
`./Build.pl build install`

xross, 2021-11-03:
I have made a change such that xcov is not a fixture but in the env (matching the enabletracing option)

closed 2021-11-03 by xross