xmos/lib_xud #295: Coverage metrics do are not calculated over the whole codebase

xross, 2021-11-10:
Coverage metrics are only calculated over code that makes it into the application test binaries.  This means that one could achieve 100% coverage by simply not using large regions of code.

xross, 2021-11-10:
@kevinyeungxmos  first solution was to simply run all tests using -O0 hoping that symbols are not removed by optimisations - however, this may mean that tests do not pass at all. Also the mapper may still throw stuff away.

xross, 2021-11-11:
@kevinyeungxmos, I took a look at the mapper options (xmap) it seems you can use the --retain option (Retain unreferenced objects). 

You can enable this by adding the following to the test_makefile.mak:

`-Xmapper --retain`

kevinyeungxmos, 2021-11-12:
I got the following error when adding `-Xmapper --retain` to test_makefile.mak
/home/kevin/sandboxes/lib_xud_DEV_XS3/lib_xud/lib_xud/src/core/XUD_TestMode.S:(.text+0x34): Error: byte offset 0x00000032 for relocation R_XCORE1_REL16_4 is misaligned
/home/kevin/sandboxes/lib_xud_DEV_XS3/lib_xud/lib_xud/src/core/XUD_TestMode.S: Error: relocation with respect to 'InvalidTestToken'
/home/kevin/sandboxes/lib_xud_DEV_XS3/lib_xud/lib_xud/src/core/XUD_TestMode.S:(.text+0x40): Error: byte offset 0xffffffce for relocation R_XCORE1_REL16_4 is misaligned
/home/kevin/sandboxes/lib_xud_DEV_XS3/lib_xud/lib_xud/src/core/XUD_TestMode.S: Error: relocation with respect to 'XUD_UsbTestSE0_loop'
/home/kevin/sandboxes/lib_xud_DEV_XS3/lib_xud/lib_xud/src/core/XUD_TestMode.S:(.text+0x64): Error: byte offset 0xffffffaa for relocation R_XCORE1_REL16_4 is misaligned
/home/kevin/sandboxes/lib_xud_DEV_XS3/lib_xud/lib_xud/src/core/XUD_TestMode.S: Error: relocation with respect to 'XUD_UsbTestSE0_loop'
/home/kevin/sandboxes/lib_xud_DEV_XS3/lib_xud/lib_xud/src/core/XUD_TestMode.S:(.text+0x6c): Error: byte offset 0xfffffffa for relocation R_XCORE1_REL16_4 is misaligned
/home/kevin/sandboxes/lib_xud_DEV_XS3/lib_xud/lib_xud/src/core/XUD_TestMode.S: Error: relocation with respect to 'InvalidTestToken'
/home/kevin/sandboxes/lib_xud_DEV_XS3/lib_xud/lib_xud/src/core/XUD_TestMode.S:(.text+0x74): Error: byte offset 0xffffff9a for relocation R_XCORE1_REL16_4 is misaligned
/home/kevin/sandboxes/lib_xud_DEV_XS3/lib_xud/lib_xud/src/core/XUD_TestMode.S: Error: relocation with respect to 'XUD_UsbTestSE0_loop'

kevinyeungxmos, 2021-12-10:
see #296

commit b340d33b (2021-12-20): fixed #295 (#304)

* Delete redundant code
* Changed -g0 to -g for XUD_IoLoop.S
closed 2021-12-22 by xross