xmos/lib_xud #298: Short EP0 packets not being ACKed in XVF3610/5

ed-xmos, 2021-11-16:
This relates to version 2.0.1 of XUD being used in v5.3.0 of XVF3610/5. This design runs at 589MHz with all 8 threads active on both tiles. 
This issue came to be noticed due to the USB volume control test failure which is documented here:
https://github.com/xmos/sw_xvf3610/issues/158

ed-xmos, 2021-11-17:
I have added @xross 's suggested debug trap in the CRC check and enumerated the XVF3610 v5.3.0. The trace files are attached below. It indeed looks like a short OUT from the host is the trigger for the CRC fail. The 

The mod in XUD_Token_Out_DI.S
```
.align FUNCTION_ALIGNMENT
.skip 0
DoOutNonIso:
    bl         doRXData
    {clre;     eq         r1, r6, r11}          // Check for good CRC16
    ecallf r1 //Trap if bad CRC
```

```
Dump of assembler code for function DoOutNonIso:
0x000933f0 <DoOutNonIso+0>:	bl (lu10)       0x30
0x000933f4 <DoOutNonIso+4>:	clre (0r)       
0x000933f6 <DoOutNonIso+6>:	eq (3r)         r1, r6, r11
0x000933f8 <DoOutNonIso+8>:	ecallf (1r)     r1
0x000933fa <DoOutNonIso+10>:	nop (0r)        
End of assembler dump.
```

And registers:

```
r0             0x80100	524544
r1             0x0	0
r2             0x80000	524288
r3             0xa10e0	659680
r4             0x0	0
r5             0x9bd80	638336
r6             0x3ffe	16382
r7             0x0	0
r8             0x0	0
r9             0xa001	40961
r10            0x0	0
r11            0x7000	28672
cp             0x97e40	622144
dp             0x9a038	630840
sp             0xe0ef8	921336
lr             0x933f4	603124	 DoOutNonIso + 4
pc             0x933f8	603128	 DoOutNonIso + 8
sr             0x490	1168
spc            0x933f8	603128	 DoOutNonIso + 8
ssr            0x582	1410
et             0x18	24
ed             0x0	0
sed            0x10400	66560
kep            0x80080	524416
ksp            0x800a4	524452
vec_vsr        0x0	0
```

So we can see R6 != R11 so this is a CRC fail which I think happened after the H2D 2 byte OUT.

[image]

[3610_enumerate.zip](https://github.com/xmos/lib_xud/files/7553660/3610_enumerate.zip)

xross, 2021-11-17:
We expect this packet to go through the "tail 0" case - 4 byte packet including 32 byte CRC. 
Expected CRC for this should be 0x7000 (as seen in R11) - actual CRC however actually matched tail2 or tail3 case.  

I believe the last CRC32 didn't have time to run - i believe this to be addresses as part of #270. This work is yet to be merged.

xross, 2021-11-17:
> @ed-xmos is going to run the test on the previous test to see if this is a regression of a long standing issue.

@ed-xmos confirmed sam issue exists on 5.2.0 release of XVF3610 - so this is a test escape

xross, 2022-01-17:
Issue resolved with latest lib_xud