xmos/lib_xud #307: Some assembly files are still missing from the coverage report

xross, 2021-12-22:
@kevinyeungxmos reports that xaddr2line returns a bad path

kevinyeungxmos, 2021-12-22:
xaddr2line -p -f -e image_n0c0.elf 0x00082740
it returns:
XUD_IN_NotReady at /home/kevin/sandboxes/lib_xud_DEV_XS3/lib_xud/tests/test_bulk_rx_basic/.build_xs3_600_6_1_1_HS/./included/XUD_Token_In_DI.S:26

xross, 2021-12-22:
Does moving the .S files from core/included to /core (and changing the #includes in XUD_IoLoop.S) help at all?

kevinyeungxmos, 2021-12-22:
> Does moving the .S files from core/included to /core (and changing the #includes in XUD_IoLoop.S) help at all?
No, it doesn't work

xross, 2021-12-22:
Looks like the ELF doesn't contain correct debug info for this #included files

xross, 2022-01-05:
@kevinyeungxmos has added a workaround (search for correct path) in xmos/test_support

xross, 2022-01-05:
In the elf/xe file the "working" files have the correct full path e.g:
/Users/ross/sandboxes/sb_xud_view/lib_xud/lib_xud/src/core/XUD_IoLoop.S

Whereas the included files just have the relative path e.g:
./included/XUD_Token_In_DI.S

xross, 2022-01-17:
Internal issue raised with XMOS tools team - Bug #18615 (http://bugzilla/show_bug.cgi?id=18615)

closed 2022-01-24 by xross