xmos/lib_xud #335: ET_ILLEGAL_RESOURCE when connecting device to Linux host

lucianomartin, 2022-02-23:
An exception is raised when a device is connected to a Linux host. The full error is below: 

```
xrun: Program received signal ET_ILLEGAL_RESOURCE, Resource exception.
[Switching to tile[0] core[5]]
XUD_GetSetupData (e=661688, buffer=<value optimized out>, datalength=0xe1594) at ../../../../lib_xud/lib_xud/src/user/client/XUD_EpFunctions.c:221

221 asm volatile("inct %0, res[%1]" : "=r"(lengthTail) : "r"(ep->client_chanend));
Current language: auto; currently minimal
```

One example of the failure in Jenkins is below:
http://srv-bri-jcim0:8080/blue/organizations/jenkins/XMOS%2Fsw_xvf3510/detail/develop/956/pipeline/1593/

The problem was introduced by the commit:
https://github.com/xmos/lib_xud/commit/c486d430f03fab93db0ad9bf95741799ada35004

Probably the simplest solution is to revert the commit above.

xross, 2022-02-24:
This traps as when converting the function XUD_GetSetupData() from assembly to C the check for received OUT data instead of a SETUP was removed (indicated by the XUD core by it sending a control token or not). This is not expected behaviour.  The 3510 codebase hits the issue when receiving a SetRes(Feature Unit 10, Volume) request. Before patching XUD it would be good to understand why this occurs - this is currently under investigation.

xross, 2022-02-24:
lib_xua does have some default cases that should in in place but the main issue appears to be that XUD_SetStall() does not work correctly when an EP is marked as ready. Testing hole: https://github.com/xmos/lib_xud/issues/337

xross, 2022-03-03:
Root cause issue is #339