xmos/lib_xud #336: ET_LOAD_STORE raised in AN00132_image_class example when device is connected  to a RaspberryPi

lucianomartin, 2022-02-23:
When running the AN00132_image_class example on a RaspberryPi, the following exception is raised:

```
xrun: Program received signal ET_LOAD_STORE, Memory access exception.
      [Switching to tile[0] core[2]]
      XUD_IN_TxHandshake () at ./included/XUD_Token_In_DI.S:36
      36    ./included/XUD_Token_In_DI.S: No such file or directory.
          in ./included/XUD_Token_In_DI.S
      Current language:  auto; currently asm
```

The exception is not present when using a MAC

xross, 2022-03-03:
Reproduced on xCORE.AI and xCORE 200

xross, 2022-03-03:
It looks like this example has a fair few bugs in it and the issue is not the fault of the core library, though perhaps it could do more sanity checking of the user input.
- the endpoint count doesn't match the descriptors. Therefore some of the tables are not properly setup and the device traps.
- endpoint type table is wrong (two control endpoints). 
- reset command from host handled as a bus reset
- quick inspection shows the data packets do not look correct from the host

xross, 2022-03-09:
Fixe in #343 also added test

closed 2022-03-09 by xross