xmos/lib_xud #349: API additions to support removal of duplicated code in the SDK

keithm-xmos, 2022-05-04:
To support integration with TinyUSB, the following functions are implemented in the SDK:

```
XUD_Result_t xud_data_get_start(XUD_ep ep, uint8_t *buffer);
XUD_Result_t xud_data_get_check(chanend_t c, unsigned *length, int *is_setup);
XUD_Result_t xud_data_get_finish(XUD_ep ep);
XUD_Result_t xud_setup_data_get_finish(XUD_ep ep);
XUD_Result_t xud_data_set_finish(chanend_t c, XUD_ep ep);
```

These functions are implemented in the SDK here: 
https://github.com/xmos/xcore_sdk/blob/develop/modules/rtos/drivers/usb/src/xud_xfer_data.h

The implementation details of these functions is duplicated ASM code copied from lib_xud.  We could remove this duplicated code if the XUD API were extended to include equivalents of these five functions.  

Related to: https://github.com/xmos/lib_xud/issues/128

xross, 2022-05-05:
I had a quick scan and I *think* everything required is in here: 

https://github.com/xmos/lib_xud/blob/develop/lib_xud/src/user/client/XUD_EpFunctions.c 

Please do let me know if there are gaps.

keithm-xmos, 2022-05-05:
Mentioning @xmos-jmccarthy so he can review.

keithm-xmos, 2022-05-05:
Closing due to the missing APIs being present.

closed 2022-05-05 by keithm-xmos