xmos/lib_xud #367: TX_FILT_ticks needs to be adjusted

henkmuller, 2023-01-13:
One of the flag_ports is clocked off the reference clock, the other is clocked off the USB clock; the latter should not use TX_FILT_ticks but (TX_FILT_us * 60).

https://github.com/xmos/lib_xud/blob/126e2837d184acae0ffe97ff148676ca9857400d/lib_xud/src/core/XUD_DeviceAttach.xc#L116

xross, 2023-01-18:
The configuration of flag port 1 was changed such that it was clocked from the USB clock (was reference) clock here: 
https://github.com/xmos/lib_xud/commit/84a8e04323b74bdadf11670a068c5633fe123dcd

This was to aid timing of RXA on xCORE, neglecting its use for signalling. We should change the config of the port as we change to and from "data transfer" and "signalling" mode.

xross, 2023-03-31:
There is a fix for this in https://github.com/xmos/lib_xud/pull/370. However, on reflection I do not think this is a good fix since it makes reference to XS1_CLKBLK_REF - any other users of this clockblock (from XC) will get an error from the compiler.  

We could use another trick to reset to the use the CLKBLK_REF, set_port_use_on() or some inline asm but I think the original suggestion from @henkmuller would be simpler. We just need to remembers that the change needs to take place on different ports on xs2 vs other devices

xross, 2023-04-03:
I recalled why I took this approach.. when the USB phy is suspended (not yet implemented) there will be no clock out from the phy so we need to take an approach similar to this in the future anyway.

closed 2023-05-30 by xross