xmos/lib_xud #381: High-bandwidth endpoint?

michele-perrone, 2023-07-07:
Hi, 

I was reading through the XMOS forum and I found the information that `lib_i2s` currently supports [up to 10 channels at 192kHz/32bits](https://www.xcore.com/viewtopic.php?t=8283), which confirms Table 4 of the [USB UAC2 guide](https://www.xmos.ai/download/Why-do-you-need-USB-Audio-Class-2?(1.0).pdf):

[image]

I was wondering: are there any plans for implementing high-bandwidth endpoints in `lib_xud`? That would bring more than triple the bandwidth compared to what we have now, enabling a lot more I/O channels for `lib_i2s`.

Thanks!\
Michele

michele-perrone, 2023-07-31:
Hello Ross, many thanks for you feedback.

At Weiss Engineering we are investigating the possibility of using X-MOS chips for the next generation of our audio interfaces in the pro-audio segment. However, the current channel count is a limiting factor in comparison to what Firewire had to offer. For example, our [AFI1](https://weiss.ch/products/pro-audio/afi1/) supports 24 in/24 out at 44.1 and 48 kHz, 16 in/16 out at 88.2 and 96 kHz, and 8 in/8 out at 176.4 and 192 kHz. In order to reach the same channel count, high-bandwidth endpoints would be needed.

Kind regards,
Michele

xross, 2023-07-31:
Michele, assuming 32bit audio and an asynchronous device the following are the maximum channel counts that can be archived with standard endpoints:
48kHz: 42 in/42 out
96kHz: 19 in/19 out
192kHz: 10 in/10 out

There may, of course, be other limitations the reduce channel count other than USB bandwidth (I/o count, MIPS to deliver I2S etc)

michele-perrone, 2023-07-31:
Hi Ross, maybe I didn't understand the table above well, or I didn't express myself very clearly.
Probably I should have written:

44.1 and 48 kHz: 48 channels (24 in + 24 out)
88.2 and 96 kHz: 32 channels (16 in + 16 out)
176.4 and 192 kHz: 16 channels (8 in + 8 out)

Your calculation suggests that at 192kHz, there is a total of 10 channels available: this means, for example, 5 in + 5 out.
Is this correct?

xross, 2023-07-31:
Ah, that table is incorrect, 10 in AND 10 out is the maximum at 192. It's one of the default configs on our multichannel audio board. I'll see if I can get that document fixed

fabriceo, 2023-11-29:
Hello, I d like to reinforce this enhancement request. 
There is more and more appetite for DSD256 and PCM384k which is not possible for 8 channels USB DAC due to the 64MBit limitation. So it is not possible to feed a beast like ES9038 with its maximum capabilities.
Using both PIDDATA0 & PIDDATA1 is needed to overcome this.
this will probably require a specific chan-end for each DATA0 and DATA1 flow and two separate buffer tasks to accommodate the required mips (guessing)...
also it is unknown if Thesycon driver and Linux/Macos driver can deal with this (either DSD256x8 or 384Kx8 or 192kx16 outs + ins).

fabriceo, 2025-07-15:
Very very good stuff. Thanks.
The information in Jira may not be publicly visible.
800mhz requirement remove the xu316-tq128 from the candidates... any chance to support HBW at 600 with slight overclocking ?

xross, 2025-07-15:
> Very very good stuff. Thanks. The information in Jira may not be publicly visible. 800mhz requirement remove the xu316-tq128 from the candidates... any chance to support HBW at 600 with slight overclocking ?

The Jira link is internal documentation (it shouldn't be linked really). The user documentation has been updated in line. 

We're working on an answer to that (which ideally isn't using a different package) Overclocking the TQFP part to that degree is NOT recommended (there is a reason why 800 parts are currently not available..). They WILL fail with high percentage. 

800 operation is required to pass test.