xmos/lib_xud #395: Setup packet occasionally hangs in lib_xud 2.2.2

Zdenek151, 2024-06-24:
Dear all,

I am seeking advice regarding the following problem, which is possibly relevant to lib_xud.
In our product, we are using xu216 with modified sw_usb_audio 6.15.2 and lib_xud 2.2.2. We received reports of our control software loosing connection with our device. The usb trace shows error in setup packets of the control transfers exchanged between the control software and the device.
[image]
The processor core state appears to be stuck in XUD_GetBuffer_Finish.
[XMOS DUMPSTATE.txt](https://github.com/user-attachments/files/15950683/XMOS.DUMPSTATE.txt)
Any idea on how to tackle this?

This issue was very hard to reproduce and apparently only happens on specific computers.

xross, 2024-06-24:
Please could you attach a full USB trace. Thanks.

Zdenek151, 2024-06-24:
@xross The full usb trace archive is about 560MB. Here is a dropbox link. https://www.dropbox.com/scl/fi/53ds76fkk00yrvh6e6o0r/lib_xud_issue395_USB_trace.tdc.tar.gz?rlkey=9f32znypjd10f6rdqjfl0s4c7&st=1p9zd7i7&dl=0

xross, 2024-06-24:
Thanks. 

Can you confirm a couple things: 
- clock frequency device is running at
- number of cores running on the tile where lib_xud is placed

Zdenek151, 2024-06-24:
@xross Both tiles are running at 500MHz, usb is on tile[1] with 6 cores.  XUD_Manager code is assigned 100MHz using set_core_high_priority_on().

xross, 2024-06-24:
Is there a chance you can try to reproduce with the develop branch of lib_xud? A small timing tweak was recently applied but not yet releases.

Zdenek151, 2024-06-24:
Ok. We can try.

Zdenek151, 2024-06-26:
@xross It looks good after a day of stress-testing. With lib_xud 2.2.2, the issue occurred after 15 minutes. Was that the RX_RISE_DELAY change?

xross, 2024-06-26:
Okay that's good news. Yes this is the main change since 2.2.2. It's a result of going back and doing more soak testing of handshaked transactions on xs2 devices. We'll make a new release very soon.

Zdenek151, 2024-06-26:
@xross That was incredibly lucky timing for us. Thank you for the help.

Zdenek151, 2024-06-26:
Closing as resolved.

closed 2024-06-26 by Zdenek151